Photonsters/PhotonFileValidator #4: Improvement sugestions

X3msnake, 2018-07-10:
[陳彦廷 loving this, just saved me from a failed print with unsupported island. 
Love the up and down shortcut.](https://www.facebook.com/groups/AnycubicPhoton/permalink/1382395165238455/?comment_id=1382988078512497&comment_tracking=%7B%22tn%22%3A%22R%22%7D)

Feedback
-slider for layer flip
-Would like to see zoom in / out function. maybe left and right keys
-see light underlay image of the models silhouette, above the support raft. this will be tricky to implement without some user input, maybe specify the layers to combine into a single image. 
This just helps to identify which part of the model islands are unsupported.

Great Job!

X3msnake, 2018-07-10:
In animation we call that onion skinning and it usually has a slider that allows the user to set the amount of visible frame layers

[image]

bonosoft, 2018-07-12:
Zoom added in version 1.1 and Layer slider added in 1.2

3DEnthusiast, 2018-07-16:
Improvement suggestion: Can you make the "Open File" remember the last folder it was in? For me it defaults to "My Documents".

NardJ, 2018-07-17:
Maybe also a resin selector which changes curing times?

iplayfast, 2018-07-17:
Fantastic work! A feature request I would like to add is importing of files. Perhaps a gif with layers, or a directory full of images that need to be made into a photon type. Or (my favourite idea) 
SomeProgram | PhotonFileValiator -i pipe -o outputfile

X3msnake, 2018-07-17:
You can do this with photon file editor

bonosoft, 2018-07-31:
Peel the Onion aka Preview 3D is now added to the application.

NardJ, 2018-08-01:
Looks very nice!

OFMsault, 2018-09-09:
It would be very nice to have ability to draw/erase on a layer with brush. Not just with rectangular marquee. And literally draw/erase pixels but not just invert it's current state.

Also, with raising appetites, ability to fill islands or cavities in one click and ability to draw on a _specific number_ or _all_ underlaying layers at once so user could draw primitive support structures under islands or connect them to mainland with bridge of some defined height.

[image]

closed 2021-10-12 by X3msnake