Photonsters/PhotonFileValidator #5: Add loading progress bar

X3msnake, 2018-07-10:


X3msnake, 2018-07-10:
Another file not loading. Seems anything with Above 2000 layers is prone to this

[image]

bonosoft, 2018-07-11:
The error is out of memory. You can test by running it from the command line: java -Xms3g -Xmx6g -jar PhotonFileCheck.jar

I am changing the way it stores unpacked layer images, to reduce the memory use.

bonosoft, 2018-07-11:
Yes it's a java limitation (default settings). Try the new version, it does not store all layers raw, but RLE  encode the layers much like the photon file.

X3msnake, 2018-07-11:
Awsome.

Gonna make a torture file to see how many layers we can get with the new
version. Btw we should use github releases features

bonosoft, 2018-07-11:
What is the GitHub release feature? Can't seem to find anything here on the web or in the app.

bonosoft, 2018-07-11:
[image]

Tried to slice this, but after 30 minutes its only reached what seems to be 10%...

X3msnake, 2018-07-11:
it's easier to just make one with PFU

https://help.github.com/articles/creating-releases/

bonosoft, 2018-07-11:
[image]

Now it works with a 5970 layer slice ( 149.25mm at 0.025)...

X3msnake, 2018-07-11:
not bad :)

thanks

bonosoft, 2018-07-12:
Now both launchers (windows and macOS) defaults to min 1G and max 5G so they should be able to load even complex max height models 155mm with lowest specified layer height 0.025 mm. Error messages from the loader thread will now be shown in the margin label.

closed 2018-07-12 by bonosoft

X3msnake, 2018-07-12:
Awsome work Master. thnaks