Photonsters/PhotonFileValidator #7: Rename this project to Photon File Validator?

X3msnake, 2018-07-11:
@bonosoft 

I think Validator or Validation makes more sense in showing the strong point of this project. @NardJ has wisely renamed the other project to Photon File Editor :) 

What do you think of the sugestion?

bonosoft, 2018-07-12:
Good suggestion, as you are probably viewing it to validate it... 

The Jar file still have the first name, Photon File Checker... :-)

X3msnake, 2018-07-12:
Checker is a good name too.

I defer to you on the chosen name ;)

NardJ, 2018-07-12:
I personally like Validator. It sounds like you really need to run this before doing your multi hour print. (Which is correct in my opinion)

X3msnake, 2018-07-12:
Awsome, let's do it :)

bonosoft, 2018-07-12:
Project, installers, launchers and application changed from viewer to validator...

closed 2018-07-12 by bonosoft