Photonsters/PhotonFileValidator #18: Request for Estimated Resin Cost to Print

MofoMojo, 2018-09-19:
The Photon File Validator does an admirable job of estimated print time and print volume (time is more accurate than the slicer!!). It would be cool to have an option to put in how much a bottle of resin was, the volume of resin in that bottle and then have it estimate the cost to print.  I currently do this math based on the volume estimates that the validator provides.

X3msnake, 2018-09-19:
you need more than the volume.
you need the weight of the bottle and the density.

volume changes on curing with material contraction. only weight stays the
same

X3msnake, 2018-12-25:
@bonosoft do you think this feature is usefull or can we close this issue?