Photonsters/PhotonFileValidator #22: Better editor for the layers - Feature Request

X3msnake, 2019-01-12:
@Photonsters/photonsters 
Here's some code we may be able to use to improve our current pixel editors, maybe we can integrate them or just add some features from it, both sources are JavaScript

- [Poxi Video] / [Online App](http://app.poxi.rocks/) / [Poxi Github](https://github.com/maierfelix/poxi/tree/master/src)
- [Piskel Video](https://youtu.be/rMal3q5Hv8I?t=47) / [Online App](https://www.piskelapp.com/p/agxzfnBpc2tlbC1hcHByEwsSBlBpc2tlbBiAgKC4jKOUCAw/edit) / [Piskel Github](https://github.com/piskelapp/piskel)

NardJ, 2019-01-12:
I am currently taking a look at voxel editors. The can serve to view a photon file (after relative simple file format conversion) and of course let's edit the layers. [Goxel](https://github.com/guillaumechereau/goxel) and [MagicaVoxel](https://ephtracy.github.io/) seem to offer large voxel scene dimensions.

X3msnake, 2019-01-13:
i did that research on this PFE issue
https://github.com/Photonsters/PhotonFileEditor/issues/83

hope you are using it as a reference and that you add new stuff you find. 

unfortunately i failed to find any voxel editor that was able to open and edit the mass amount of data we are talking about here. let alone open source versions.

only real solution would be to use a closed source advanced voxel engine like atomontage, but that is out of question.

alternative is to use some open source editor as a start and develop a better compression algorithm on memory to show and edit and generally move around the rebuilt voxel model.

i think besides octree tye best solution in the academic circles i found is either fractal data compression. i believe i have dropped links to these on that open issue too

dailyrebel, 2019-11-11:
I would love to have this function. It takes me ages to delete the pixels in hundreds of layers manually

closed 2021-10-12 by X3msnake