Photonsters/PhotonFileValidator #24: Request support for new PhotonS files

littleneutrino, 2019-03-28:
Can we get support for the new PhotonS files, I tried just renaming the file to Photon and putting it through the program but the validator says the file is corrupt. so they must have made some sort of  change to it.

bonosoft, 2019-03-29:
The PhotonS is a completely new machine, new hardware and file format.

If AnyCubic is willing to provide information on the new file format, it is a doable task - else reverse engineering the file format will take some time and require that someone is willing to put in the hours.

greenwoodma, 2019-04-05:
@bonosoft I saw that you recently committed code to handle a v2 file format. Is this for the PhotonS or do any of the extra settings apply to the original Photon as well? Having just saved a file as v2 and tried it on my original Photon it seems to work (I didn't have any resin in so could see the layers being shown correctly) but I don't know if it's just ignoring the new values or not. Any ideas?

bonosoft, 2019-04-05:
Hi

The Validator does not support Photon S as we don’t know the file format.

The version 2 file is an extension of the version 1 format, so printers like the Photon can print both version 1 and 2 files. Version 2 files can hold extra parameters and AA information.

The firmware in the Photon currently only reads version 1 information out of the files. AnyCubic promised to release a new firmware for the Photon that supports version 2 files, but we are still waiting.

Another vendor, which have a printer based on the same ChiTu main board, have a firmware that supports version 2 files. If you install this firmware on your Photon, it will be able to print AA version 2 files. But that will void your warranty.

/Bo Nørgaard

greenwoodma, 2019-04-05:
Thanks, makes perfect sense and we'll just have to hope AnyCubic come up with some official firmware at some point.

Perhaps oddly it was actually the lift speed/height I was interested in and not the AA option as I've been having issues with resin trapped around the supports and so I thought I'd have a play with the extra settings to see if I could make things any better.

X3msnake, 2019-04-08:
@greenwoodma @littleneutrino 

Don't wait up for Anycubic to decide to put out the file format, write to them demanding backwards compatibility with the old file format or release of the binary format.

And convince others to do the same. if everyone presures Anycubic there is a better chance they do make something otherwise nothing will happen

X3msnake, 2019-04-08:
@greenwoodma lift and speed is even worse, you can do that on Photon Classic because we got a hold of the gcode file that allow these values to be changed, we don't even know what kind of implementation Anycubic decided to make on their S model

greenwoodma, 2019-04-08:
@X3msnake yes, I'm going to ask them some questions about the Photon Classic files and firmware. Obviously I'm going to ask about AA as we know that's possible but what I'd also like is to see if it's possible for the printer to respect the per-layer values for exposure, off time etc. That would be really useful as it would allow us to change layer height for different sections of a single print; i.e. you could have the raft and supports at 0.05mm and the detailed model at 0.02mm which would speed up printing.

bookkojot, 2019-06-13:
I reversed format, but later found reader here:
https://github.com/hudbrog/xPhotonS/blob/master/photons_reader.py

My work:
```
/*

0 int32_t version? multiplier for offset?
4 int16_t val=*2; preview offset? header size?
6 double pixel_size
14 double tickness
22 double normal_exposure_time
30 double off_time;
38 double bottom_exposure
46 uint32_t bottom_layers
50 double z_lift_distance
58 double z_lift_speed
66 double z_retract_speed;
74 double VOLUME
82 uint32_t preview_width
86 uint32_t preview_offset_top?
90 uint32_t preview_height
94 uint32_t preview_offset_left?
98 ...image data... uncompressed 16-bit bgr565 preview_width*preview_height*2bytes
...
...uint32_t slices count
... ...slices...

slice:
0 uint32_t white_pixels;
4 uint32_t left; (0)
8 uint32_t top; (0)
12 uint32_t width; (1440)
16 uint32_t height; (2560)
20 uint32_t (size>>3)-4
24 uint32_t width&height (reverse bits) = a0 05 50 00 -> (5a0 a00)
28 ... rle-packed image ...
Format same as original photon, but bits in reversed order

A00=101000000000 -> reverse
->  000000000101 = 0x5

0xFE -> 0x7F

*/
```

New format even simple than original *.photon

Teifun2, 2019-09-30:
I was wondering if there was work done on the new file format now that some sort of format is known?
Btw. i would add some message, that the file is not supported when opening an unsupported file. It took me some seconds to understand what the problem was.

How easy is the code source to contribute? I could try and implement this but i am unsure how well the codebase is designed to support different file types?

talpatlan, 2019-12-28:
Any hope for Photon S file format support now?

X3msnake, 2020-01-27:
Only if someone picks up the code and add it, since Bo Noordgard (the main
coder) has been MIA for the past year.

à(s) 13:53:

TobseF, 2020-04-08:
Here is another parser available for the Photon S `.pws`-format:
https://github.com/Frans-Willem/sla-format-tools/blob/master/src/formats/photons/data.rs
This [Rust](https://www.rust-lang.org/) lib can also convert between different SLA printer formats (Prusa SL1, Photon, Photon S).

billrobertson42, 2020-04-11:
Thanks for the info @TobseF 

I've been looking at adding that this evening, just for the sake of convenience. I found a fork https://github.com/cwoac/PhotonFileValidator that seems to have some of it done, but not quite all.

I also managed to track down some info from the following locations (not all seems to agree)

* https://github.com/Ordenritterlein/PhotonsFileConverter/blob/master/JS/photonsfileHandling.js
* https://github.com/robfitz273k/PhotonS/tree/master/PWS
* https://github.com/Photonsters/anycubic-photon-docs/blob/dev/Temp-resources/2019-03-05%20photons%202.0%20file%20format.pdf

The second and third seem to be the farthest out from what I see in the file (as generated by chitubox) e.g. there's no textual bits the lead the header.

serhattsnmz, 2020-11-07:
Is there any improvement about supporting new .photons files?

X3msnake, 2020-11-08:
Validator wil not be developed any more

use UVTOOLS instead

closed 2021-10-12 by X3msnake