Photonsters/PhotonFileValidator #25: Feature Request: Ability to delete all islands on all layers automatically

nickaccent, 2019-04-23:
It would be lovely to have the option to delete all islands from all layers (for when there are lots even after running the fixer, 90% of time i just end up deleting them manually on each layer affected anyway). 

This could be implemented as an optional step after the fixer has run.

SixFiveLive, 2019-05-01:
I think deleting all islands would lead to missing problem areas and possibly failed prints, or a lot of resin cured on the FEP. Just deleting lone pixels would save a lot of time going through layers without as much risk though.

rjp, 2019-10-12:
If you have a lone pixel island **with nothing on the next layer up**, I think that's a valid candidate for a secondary "Fix isolated pixels" option. Like @nickaccent, I spend 90% of my time deleting single pixel islands that Chitubox inexplicably creates.

dailyrebel, 2019-11-14:
I would love to have this option 👍

rjp, 2019-12-13:
Or if you have a lone pixel island with nothing underneath it and coverage above, that too can be automatically deleted.

Eirenliel, 2020-02-20:
https://github.com/Photonsters/PhotonFileValidator/pull/54 implements this. It's easy to change to "remove single-pixel island" if needed.

riggiding, 2020-02-24:
you can also remove all islands of the current shown layer using alt+click, next layer is automatically revalidated.

Eris-Bureau-of-Strife, 2021-09-12:
Now as for unsupported layers you are just going to have to fix them manually. But once fixed,  for true islands in 3D space I use a macro recorder and open a pixel zoom window in photon validator. I hold down alt the entire time, alt+Lclick to close the pixel zoom window, alt+Lclick to close where the "OK" button just was (more sturdy macro that way), alt+Lclick in the top right corner to clear red pixel islands then click right arrow to go to next layer. Stop recording the macro, then play it on a loop until all 15,000 layers are fixed.