Photonsters/PhotonFileValidator #27: Files stuck at loading

tdub415, 2019-05-15:
When I open some files (typically larger files that were fairly full plates in the x and y dimensions sliced at .02 height), the validator gets stuck at "Loading file" and never actually gets to the point at which I can preview the layers or search for islands.

uglyduckling81, 2019-05-26:
AA enabled?
Pretty sure File Validator can't open prints with greater than 2xAA from me experience so far.

xanrag, 2019-07-24:
I also have this problem and I do not have AA enabled. I could find no reason why some files work fine and some do not, so finally I downloaded the source and got it working in debug mode.

I was very surprised to see that the version I ran in debug mode and stepped through the code loaded the file without any problems without any changes, but the precompiled version I downloaded from here does not.

I don't know where the problem is except it is between the lines "Loading File" and "Complete" in PhotonLoadWorker.java where it fails without throwing an exception in the precompiled version.

It was a real PITA to get it to compile since I have not really worked with Java before and it required some specific software and libraries but there was no information. I finally got it to work with the IntelliJ IDEA editor, OpenJDK 12.0.2 JDK, OpenJFX 12.0.1 and applejavaextensions-1.4.jar. Presumably something in that list is newer than what was used to compile this and thus makes it work.

eljaydub, 2019-09-23:
I have this problem as well - At some point while "Calculating photon file layer" the program gets stuck and will never progress. I have AA turned off. For example right now I am trying to open a 44mb .photon file with no AA and it keeps getting stuck at layer 1255 of 1499.

eljaydub, 2019-09-23:
The previous file was sliced at 0.025mm per layer. When I re-slice at 0.04mm per layer I the get the file size down to 28mb and 936 total layers and everything works fine. I suspect this error has to go with the program running out of memory. Specifically, perhaps the Java virtual machine is not allocating enough memory for PFV?

xanrag, 2019-09-24:
> Here is the file that keeps getting stuck
> https://www.dropbox.com/s/ck2y8f4g03isrul/ghoul-body.photon?dl=0

This file opened on my regular copy of photon validator without any issues and did not get stuck. All my issues were either that the file worked completely or not at all, I did not have any problems with it stopping mid-calculation. 32bit JRE.

eljaydub, 2019-09-27:
Strange, I'd like to know what the differences between our copies of PFV are. Here is another (large) file that also never goes past loading
https://www.dropbox.com/s/fj1xn9a2s8vhj48/turbolaser_gun-body.photon?dl=0

xanrag, 2019-09-27:
> Strange, I'd like to know what the differences between our copies of PFV are. Here is another (large) file that also never goes past loading

I can confirm this does not go past loading on my regular PFV. However, if I open the source code of the same version in the IntelliJ IDEA compiler (as described above) it opens the file without any issues. Thus the issue is not with the source code itself but the compiled .exe file.

parrita710, 2019-12-10:
I get this error when executed as admin in CMD.

C:\WINDOWS\system32>java -jar C:\Users\\Downloads\PhotonFileValidator\PhotonFileValidator.jar
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
        at photon.file.PhotonFile.getWidth(PhotonFile.java:286)
        at photon.application.base.BaseForm.showFileInformation(BaseForm.java:189)
        at photon.application.utilities.PhotonLoadWorker.done(PhotonLoadWorker.java:81)
        at javax.swing.SwingWorker$5.run(Unknown Source)
        at javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(Unknown Source)
        at sun.swing.AccumulativeRunnable.run(Unknown Source)
        at javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(Unknown Source)
        at javax.swing.Timer.fireActionPerformed(Unknown Source)
        at javax.swing.Timer$DoPostEvent.run(Unknown Source)
        at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)

riggiding, 2020-02-12:
Had a look at the files provided here - they all work fine with the current version

X3msnake, 2020-02-12:
That is awsome news

à(s) 10:01:

hecksagon, 2020-06-20:
OpenJDK using OpenJ9 appears to work much better than Java JRE. I have no problem opening large files or high AA levels.

https://adoptopenjdk.net/releases.html?variant=openjdk14&jvmVariant=openj9

metaclay, 2021-06-03:
Hi, i just installed mac version. I tried to make couple of simple .photon files. When i import to photon file validator, nothing happen, no calculation progress or  whatsoever . It just go straight to 'complete' status but i got nothing - it's just blank, can't preview anything. The .photon files are just simple test i created using chitubox or photon workshop. No matter it is , whatever i throw into photon file validator, i always got blank , no progress , only the 'complete' status which instantly appear when i load .photon. What is wrong ?