Photonsters/PhotonFileValidator #32: Documentation on "Convert"

theC4Labs, 2019-08-21:
I am impressed by the software you made.

I am making some precision parts (planetary gear) and  have discovered that the LCD printer makes parts two large by ~.1mm in  XY direction. 

I am looking for a way to decrease the edge pixel(s) of each layer to compensate for the overspray error of the light. I am hoping the matix in the convert option of your software will give me this ability to remove edge pixels.

X3msnake, 2019-08-22:
No. Validator is used for validation only.

You can slice with PRUSA SL1 with XY compensaton and use SL1 to Photon
conversion tool.

There are also other tools like VoxelOps and another tool listed on
photonsters.org main page under community projects that do this. You can
open the file with Photon File Editor dump the file to images and then use
erode batch with photoshop or gimp

The .photon format is readable by Any CBD printer Mars,EPAX, Orbeat...

à(s) 21:24:

X3msnake, 2019-08-22:
Alternatively Z-Suite from Zortrax has just added support for cbddlp

albeit the format is the same you will need to rename it to photon if you are running a offcial firmware. If you however insatll EPAX 4.2.18 you will be able to read both cbddlp and .photon alike

[image]

theC4Labs, 2019-08-22:
Thank you so much for the help!

Z-Suite is exactly what I need for making precision parts.

For the benefit of anyone looking for the same solution I tell how I
finally managed to use Z-Suite to make a file for my printer.
1) Download it - it's free!
2) On the welcome screen select "External LCD Printers" which is small text
above all the thumbnails of the company's own printer models. This will
allow you to specify bed size/DPI and also the export format (Cws, Zip,
Photon, Sl1, Cbddlp).
3) The software guides you through a linear process with next buttons and a
progress bar at the bottom of the screen. The second step is "Print
Settings." To the right will be a menu - click the "Advanced" tab and you
will see "Overgrowth Correction" halfway down.

closed 2020-02-11 by X3msnake