Photonsters/PhotonFileValidator #40: It doesnt work with Chitu Box 1.6.1

dailyrebel, 2019-12-02:
The Validator doesnt work with Chitu Box 1.6.1 anymore. Please update

mprenter, 2019-12-04:
Just to expand on this, apparently .cdbblp files created by Chitubox 1.6 or 1.61, opened by PhotonFileValidator, repaired, and saved out, no longer print correctly.  The slices are solid white for the entire build plate.

johanneslumpe, 2019-12-05:
Definitely would be awesome to get an update for 1.6. The ability to fix minor islands manually in this tool is invaluable @bonosoft - props for your hard work!

Jorenm, 2020-01-30:
Can confirm. The files look fine when loaded into a slicer after being processed, but they print mostly solid bricks.

X3msnake, 2020-02-11:
https://github.com/Photonsters/PhotonFileValidator/pull/46

Can you guys test this Push branch and tell me if it fixes this issue for you

riggiding, 2020-02-11:
Maybe I should mention that I didn't build the binary - I'll update the jar so you don't need build for testing.

X3msnake, 2020-02-11:
ok thanks

riggiding, 2020-02-20:
the fix has been merged

mprenter, 2020-02-21:
Is this working for anyone else?  I'm running the new jar file and mine stops during the file import.  It says "Calculating photon file layer" and counts up to 917/1758 and just sits there with no cpu activity.

iKemp, 2020-02-21:
Have you tried to give it more memory? Had this behaviour once with a full build plate.
java -Xmx1024m -jar '.\PhotonFileValidator.jar'.

riggiding, 2020-02-21:
Not sure the jar is up to date - a build might be necessary

mprenter, 2020-02-22:
iKemp was right, it needed more memory.  I ran with the -Xmx1024m setting and it didn't pause calculating layers anymore.  However, once I ran "fix" and edited any remaining isolated pixels, saved the file, and tried printing.  Nothing printed.  It was like it was all "black".  This was confirmed by loading the output file back into Chitubox.  Nothing in the .cbddlp file.

riggiding, 2020-02-22:
@mprenter I rebuilt the jar - it was not containing the fix for a merge problem. I updated and tested it. Is fine to go now.
You still might need the -Xmx setting if your file is large.

mprenter, 2020-02-22:
@riggiding Thanks!!  Good news, bad news.  The saving and subsequent printing seems to work now.  That's awesome.  But pixel painting is now a pixel off.  You need to select the pixel ABOVE the one you want to paint.  Not sure if I should open a new issue for this.

I'm gonna need to buy you a coffee after all this.  :-)

riggiding, 2020-02-22:
@mprenter Great! Regarding pixel editing I‘ll have a look tomorrow. Coffee always welcome!

riggiding, 2020-02-23:
@mprenter Had a look at the build but could not find any issues with pixel editing. Mind to explain in more detail please? Thx

mprenter, 2020-02-23:
Sure, I load a file and run ‘fix’. Then there are usually a few areas left that require manual editing. I jump to the problem layer and click on the red area to bring up the pixel editor. It’s at this point that the targeting of the cursor seems to be a half a pixel off. For example, if I place the cursor on a red pixel and click, the pixel below where I clicked will change. To change the desired pixel I need to actually click in the bottom 1/2 of the pixel directly above the one I want to change. It’s like the ‘mouse over’ pixel grid is 1/2 a pixel up from where it is supposed to be. Makes accurate painting difficult. Does that make sense?

riggiding, 2020-02-24:
That makes sense, but I don't see it (working on Mac using Java 1.8.0_181) :-/
Also tried non mirrored projects just to be sure.
**Whats your platform/java version?**
You got a small grey cursor in the current version - **maybe you can take a screenshot for me showing the mouse pointer in relation to this cursor when its off the most?**

mprenter, 2020-02-24:
Hi, using Java on a PC. Build 1.8.0_241-b07)

[image]

Eirenliel, 2020-02-24:
I have the same problem and the difference is even bigger on high DPI if it's not scaled (like when eclipse launches it).

riggiding, 2020-02-26:
@mprenter, @Eirenliel I just merged a probable fix and the new pencil tool. If there are still Problems regarding the mouse pointer please open a new issue.

closed 2020-02-26 by X3msnake