Photonsters/PhotonFileValidator #43: Windows Scaling

mikepetrie, 2020-01-19:
On Windows, with a high resolution screen that is scaled at greater than 100%, PhotonFileValidator (java installer) ignores the Windows display scaling, making it too small to be usable. See attached screenshot of the PhotonFileValidator in front of this bug report. The Windows display scaling is set at 200% in the screenshot.

[image]

RichardCsuk, 2020-01-29:
The work around for this is:
Go right click on the PFV icon > Properties> Compatibility> Change high DPI settings. Click Override high DPI scaling behavior.  And change the drop down "Scaling Performed by:" to System.

Eirenliel, 2020-02-27:
When launched from launch4j launcher java process inherits DPI awareness of it. It looks like launch4j doesn't have an option to change this. I tried to mess with .matifest files, but it didn't help. I'll keep looking probably...