Photonsters/PhotonFileValidator #44: Pixel Changes do nothing ?

Mortu1412, 2020-01-24:
If i check the "fixed" file in chitubox it doesn't show any changes (automated or manual). Is this just a visualisation Problem by Chitubox or are the pixels not "there"

X3msnake, 2020-01-24:
are you saving the file after you edit and before you exit the software?

Mortu1412, 2020-01-24:
Thanks for the fast answer! I did save it and i also tripple checked if i open the correct file in Chitubox and i also open the file in Validator. For the same Layers i see different Pixels. I'm going to make a vid if it helps and upload it as private to YouTube. Maybe I'm doing something really stupid, but as mentioned im pretty sure i don't.

Mortu1412, 2020-01-26:
@X3msnake i did some more research and observed that indeed the Changes are visible in the 2D-View of Photon Workshop and Chitubox but only as a greyish colour? It is not visible in the 3D-View.

X3msnake, 2020-01-27:
the 3d view uses only the first layer to build the model. same as older
firmwares will discard the AA layers and useonly the first.

if you dont have your main print in that layer it will not show on 3d model

closed 2020-02-11 by X3msnake