Photonsters/PhotonFileValidator #45: Will Photon File Validator Support Chitubox Large Size .ctb files ??

Portaler111, 2020-01-31:
So pretty much as the title says. When will the file validator support .ctb files from Chitubox for the Elegoo Mars Pro ? 

This is such a great little piece of software. Would be great to see.

X3msnake, 2020-02-01:
we are loooking for devs to implement this. bo has been MIA

Cylindric, 2020-03-07:
Yeah, this would be so nice, as the Mars Pro doesn't support the old file format at all. 
I'm curious, is the file format known? I might be able to help with the coding if it is.

porlock, 2020-04-14:
+1 on for that from me.

X3msnake, 2020-04-14:
@Cylindric like all cbd-tech formats we will probably have to do the legwork of reversing it to learn how it works. It was like that for Photon and Cbddlp i don't see xinfu xie and the chitubox team giving away the bluprints this time around either, but do try pressing them on the chitubox FB page. 
Alternative is to poke at it old style like we dif for the .photon with hex editors lots of files with known changes and comparisons. veles and binviz are a good open source starting point

porlock, 2020-04-14:
> @Cylindric like all cbd-tech formats we will probably have to do the legwork of reversing it to learn how it works. It was like that for Photon and Cbddlp i don't see xinfu xie and the chitubox team [...]

Does this *.ctb files are the same that are used for plotters ?  if yes it soul be possible to opene it by autocad.

X3msnake, 2020-04-14:
nope. not related at all.

porlock, 2020-04-15:
*.ctb files seams to looks similar to multiple images .ico files

X3msnake, 2020-04-15:
this is what we know about ctb and phz #63 

https://github.com/Photonsters/PhotonFileValidator/files/4482843/Chitubox_Files.docx

porlock, 2020-04-15:
Lol I was thinking I will be use full :) anyway it seams that you know already alot

cbiffle, 2020-04-21:
In case this would be helpful: https://github.com/cbiffle/catibo/blob/master/doc/cbddlp-ctb.adoc

Just found your doc, haven't finished cross-referencing my field definitions with yours, but I think I've got some parts you haven't (and vice versa).

X3msnake, 2020-04-21:
@cbiffle awsome findings, how did you figure out the encription?

porlock, 2020-04-21:
!! holy shit, if PFV will supports *.ctb that will be like x-mas in summer.

Greger312, 2020-04-26:
.ctb support asap and I will carry your love child.

X3msnake, 2020-05-01:
https://github.com/ddv2005/resin_xp_finder_mars_pro/blob/master/template/ctb_hextemplate.bt
https://github.com/ddv2005/resin_xp_finder_mars_pro/

porlock, 2020-05-02:
@X3msnake great finding, now it should be just a matter of coding and testing, all hacking and reverse engineering is already done. Any ideas when .ctb support will be ready :) ?

X3msnake, 2020-05-03:
not for me to say. i have no java skills. anyone is free to contribute
thought

cpd1, 2020-08-04:
Is any one working on this?
If no one is...  I’ll give it a crack.

porlock, 2020-08-04:
I think no one, so if you can, please do

X3msnake, 2020-08-05:
most of us have gone to UVTOOLS by Sn4k3
but feel free to implement new features and submit prs on this

cpd1, 2020-08-05:
Which does a better job of detecting islands?
I'll continue to work on this if it's better than UVTOOLS for island detection.

X3msnake, 2020-08-05:
at this point UVTools is a much more powerfull suite

not only you have island detection, but also trapped resin detection

it also features a better editor and has the hability to place some
simplified supports on a sliced file

only current downside is, it is not cross os compatible yet (except running
it on wine)

cpd1, 2020-08-05:
Ok. Sounds like UVtools is the way to go.  
Thanks for the info!

soswow, 2020-09-05:
UVTools is only a Windows app, though =(

dailyrebel, 2020-12-11:
This would be so great if file validator could support the ctb files

X3msnake, 2020-12-13:
use UVTOOLS instead. PFV is no longer being developed