Photonsters/PhotonFileValidator #47: Support generic zip files or image folders

X3msnake, 2020-02-11:
Would be great to be able to use with zip files

cwoac, 2020-03-15:
I'm currently working on a patch to add SL1 support, which is (basically) a zip of pngs (+ a config.ini file for settings), which I guess could be made generic enough to handle this. What were you thinking of for putting in the default settings in such a case?

X3msnake, 2020-03-15:
chitubox generic zip is the ideal candidate.
It basically is image files + Gcode + Config

cwoac, 2020-03-16:
OK. I've taken a look at that, it's _almost_ identical to the SL1 format, so adding both shouldn't be that hard.

I do need to think a bit about the right place to add the abstractions to handle the different metadata blobs the three (four if photons has variance from photon) formats require - `IFileHeader` is currently very photon specific.

Also need to consider what the use case is here, exactly - I can see three options:

1.  Loading zip files, validating them, saving them back out as zips - i.e. the roundtrip option by implementing savers and loaders with (probably) individual abstractions of `PhotonFile`
2.  Loading zip (or other type) files, validating them, saving them back out as photon - i.e. only implement loaders, and mapping everything to photon terms internally.
3.  Loading zip (or other type) files, validating them, saving back out as zip (or other type) files - i.e. implementing both savers and loaders with a common backend object.

2 would probably be the easiest (and from my pov, sufficient), but probably the least satisfactory. 1 would probably be the lightest touch in terms of refactoring, but I think 3 is the most valuable.

Just to be clear, I'm only really talking about refactoring / modifying the code for saving and loading here, I'm staying well clear of the internals.

riggiding, 2020-03-16:
I'd vote for abstracting the internals from the file format/loading/saving completely - its more work now but I'm sure there are many more formats to come.

cwoac, 2020-03-16:
OK, I've made a start on it, Nowhere near ready for PR, but the code is being pushed to cwoac/PhotonFileValidator@c8790f8de6d40c86693f30c34059cae198f5ba40

Roughly:
* I've basically renamed `PhotonFile` to `SlicedFile` (open to better names), made it abstract and added abstract save/load functions.
* I've created implementation classes for Photon, PhotonS and SL1 files currently. The Photon one is just a refactor, so works as in the current code. The PhotonS uses the stub code that I see someone had started on (header loading only so far), the SL1 code is my stuff and is about at the same point as the PhotonS stuff right now.
* I've added an enumeration for handling filetype detection and pointing the load code at the right implementation class

Yet to do:
* implement layer loading from PNG files. I should be able to lift code from the pack routines to do this though.
* implement AA layer loading from PNG files. I should be able to lift code from the AA generation routines to do this.
* saving native format
* figuring out how to convert between formats, especially given the difference in what metadata is stored (e.g. the zipfiles have a bunch of gcode commands added).
* What to do about formats which don't have previews (use first layers?)

cwoac, 2020-03-16:
For saving/conversion I see two options:
* add a new abstract function `fromSlicedFile` to `SlicedFile` - this would mean you only have to write one conversion function per format, _but_ you could only use stuff common to all formats. This should be fine for the actual layer data, but slightly limiting on metadata (e.g. I expect the metadata for Photon and PhotonS files to be very similar). You could work around this using some slightly nasty type introspection if needed though.
* Add new functions per type to each implementation. This could then maximise reused metadata, at the expense of needing a new routing workflow for saving and quite a lot of extra code.

I'm strongly leaning towards the first option.

cwoac, 2020-03-16:
Further point to consider - this approach is _not_ lossless, even with png in zips (both chitubox and sl1 files) - why? because the file uses greyscale value to represent AA values, but photon uses discrete layers. This means each value will be (in effect) rounded to the nearest AA value of the level we choose to import at.

Which brings to the next question - what AA level _do_ we choose to import at? Obviously the higher the level, the more time / memory we need to import, but we probably want to maintain some level of it. Offering the use a choice / setting would be a good solution, but that has a bunch of UX considerations - perhaps following the pattern used for margins?

For now, I'm going to go with 4 as that is the default on the photon these days.

riggiding, 2020-03-16:
@cwoac regarding the metadata I think the abstraction could provide all there is and the implementations need to enable/disable whats actually there.
kind of enum MetaData { exposure, .... }, and a function per format isSupported(MetaData.?)
does that make sense?

Regarding AA: I'm not sure this needs to be imported at all - I think the AA layers are always regenerated anyway when saved? Makes sense as they most likely need to in case anything has changed.

cwoac, 2020-03-16:
Yeah, It is kinda already there as a pattern with hasAA, and I'd just added hasPreviews.

I think the AA still needs adding - given how long the AA layers take to add when you import a file that doesn't already have it, I'm pretty sure that it isn't entirely recalculated on save.

cwoac, 2020-03-17:
OK, I've got it working to the point of being able to load and fix SL1 files (no saving yet), which is probably good enough for actually looking at the code if people care to - it's now in #65

cwoac, 2020-03-18:
OK. Saving (as photon) now works! Remaining tasks:
- zip file configs
- saving as zip / sl1
- AA

cwoac, 2020-03-19:
Zip file loading now works, so you can investigate that if you like (although only .photon and .sl1 saving currently - the run.gcode file is going to be a pain.).

X3msnake, 2020-03-19:
that is already great news.

19/03/2020 à(s) 14:30:

gmlewis, 2020-03-19:
> Zip file loading now works, so you can investigate that if you like (although only .photon and .sl1 saving currently - the run.gcode file is going to be a pain.).

This is awesome!  Thank you, @cwoac!

I may have a dumb question - but let's say I have a ZIP file containing hundreds of PNG files ( in the Z direction: `out0001.png`, `out0002.png`, ...) that have been prepared for input into  NanoDLP (which means that the resolutions are: X:65, Y:60, Z:30 microns respectively).

Could I use this program to convert this ZIP into a file format that the Elegoo Mars could import directly without needing to use NanoDLP?

If not, what should the formation of the ZIP file be for optimal printing on the Elegoo Mars?

Thanks again!

cwoac, 2020-03-19:
> This is awesome! Thank you, @cwoac!
>
> I may have a dumb question - but let's say I have a ZIP file containing hundreds of PNG files ( in the Z direction: `out0001.png`, `out0002.png`, ...) that have [...]
Yes...
The easiest way would be to make a `config.ini` SL1 style, add that to the zip then rename the zip file to end in sl1 (you _could_ keep it as a zip, but then you'd need both the preview files _and_ to create the more complex `run.gcode` file).

gmlewis, 2020-03-20:
Just FYI - I found out a way to write the `.cbddlp` and `.photon` files directly due to your help on this thread. Thank you, @cwoac!

(I'm using https://github.com/gmlewis/irmf-slicer to convert irmf.io models to voxels for the Elegoo Mars.)

cwoac, 2020-03-20:
Right, zip format saving now works - main pain was that it stores that entire actual gcode to write the file, so I had to regenerate all of that (I've eyeballed that what it saves out looks close enough to what it loads in, but will need testing by someone - I don't actually know which printer uses this format!).

Last bits are other->zip and AA ( I think) then this might actually be done!

X3msnake, 2020-03-20:
i'll see if i can test it on my Elfin this weekend.

Problem is elfin uses a custom gcode format for wich i had to write a
plugin to use with chitubox :P

Basic parsin stuff. if the gcode is ok i can probabably hack the Jar i use
to conform the output to the correct one

20/03/2020 à(s) 13:34:

cwoac, 2020-03-22:
I realised that the currently mix of fields in classes and additional parameter map was unmaintainable and a bit of a mess for format cross-conversion. Have just spent a chunk of time reworking it to something a bit like what @riggiding  suggested in that all files can potentially have all parameters as they are pulling from a uniform list of known fields, all stored in a hashmap. 

This has two major points where 'fun' occurs - where different text based files use different names (e.g. 'cost' vs 'price') and where they store different types for the same field (photon stores estimated time as an int, all the others as a float).

For now, there are few enough of the first that I've simply picked one field name (the photon one if it exists) and made a note inside `EParameters` of what the correct field to use is actually called for other file types. There are a few here that I'm not 100% on (mostly on whether that dropSpeed in the zip format is the same as retractSpeed in a photon file). 

For the purposes of format differing, I _think_ the only instance is the aforementioned print time, for which I've simply made some functions to hard force between float and int - any file format should probably call them inside the header constructor that takes a SlicedFileHeader object.

Anyway, Photon saving, loading and conversion now works with the new system. SL1 saving and loading works, Zip loading works.  Sl1 conversion _probably_ works, but I have to test it. Zip saving and conversion needs me to finish writing the code to write out all the expected parameters, which is not a task for now.

cwoac, 2020-03-31:
Update! AA loading from / saving to image files now works! Well, at least it produces things that _look_ right. Very slowly*. I'm running a test print now.

* I'm pretty sure I can speed it up by using graphics2D to just layer the images for me, at least on the png/sl1 export side; but I might save that for a follow up PR as this one is large enough as it is.

cwoac, 2020-04-01:
Test print workked! Now to tidy up the PR and make sure all the conversion sequences produce valid files (photon <> sl1 <> zip)

closed 2021-10-12 by X3msnake