Photonsters/PhotonFileValidator #58: Very fine support structures

jukefoxer, 2020-02-26:
Hi,

First of all: Thank you for creating this very useful software!

because our company needs a lot of functional 3D printed prototypes I was playing around with different slicers and support structures. My problem was: The usual support structures always left ugly marks on my models when I removed them (or even damaged my model). So I implemented my own support structure algorithm. It consists of a large number of regularly spaced support structures that contact the model only with a few pixels (4 x 4 pixels at the moment, at a distance of 0.6 mm). That way the structures can be removed more easily and they leave only a very fine, regular dot-pattern on the model. I can use a brush to remove the rest of the supports and am left with a very clean looking model (before post-curing). Check [this picture](https://drive.google.com/open?id=1FFRVE0reRv33XfKKyMOHHj6wYsWX6QSn) to see the underside of the model on the left side and the support structures on the right side. The model in the picture is 38 mm x 17 mm in size.

I currently implemented this functionality in the PhotonFileValidator, but am aware, that this might not be the ideal place/project. Basically I did so, because I could do it in Java and it seemed easy. So my question is: Would this be something that could be useful (and fit) in one of the Photonster projects? 

I was contemplating publishing this feature in some form, because I think this might be useful for other people. I guess it would be quite some work for me to port it to the right project and document the usage accordingly. So that is why I'm trying to find the right place first and then decide if I want to do the work and create a pull request.

Best,

Sam

riggiding, 2020-02-26:
I do like the idea!
But if it fits in validator is a good question ... @bonosoft, @X3msnake ?

X3msnake, 2020-02-26:
@jukefoxer this type of tools would make more sense in a software like VoxelOps or PhotonFile Editor.

That said who are we to lock down how users use the Validator. I think we should open some sort of extendability so people can develop plugins to use withing validator.

@bonosoft  @riggiding is this plugin capability something hard to develop

@jukefoxer contact us at Photonsters FB page so we can add you to the group chat

riggiding, 2020-02-26:
@X3msnake plugins are a nice idea. 
In general its quite a lot of work to supply an API that is open enough but yet simple to use...

@jukefoxer can you give us access to your fork so we can get a better grasp on what kind of API you need?

jukefoxer, 2020-02-26:
@riggiding The fork is here: https://github.com/jukefoxer/PhotonFileValidator

I agree, that a (good) plugin API sounds like a lot of work.

I haven't implemented any feature to lift the model away the plate. I guess I would have to add that in order to make it easier to use. So for now you have to use a photon file that has a model that floats 2 mm above the XY-plane and then generate the supports.

@X3msnake I contacted you on the FB group.

jukefoxer, 2020-02-26:
I just cleaned up the code a little bit, added some explanations to the support creation dialog and implemented the model lift functionality (lifting it X mm from the build plate before adding the supports). 

However, I could not make the model-lift functionality work for Photon-S files because some features are missing in the header classes (and it seems that AA is not implemented for Photon-S for similar reasons).

bonosoft, 2020-02-27:
The Photon-S file format is only partly implemented, and was stopped on request from AnyCubic.

I learned that the other projects are currently not active, so the validator is probably the only option for implementing this. And like Mario, I do like the idea.

If I understand the workflow, you would like to:

- Place you model in your favourite slicer, and create a Photon file without support.
- Open the file in the Validator
- Lift the model X number of layers
- Add remove-friendly support structure to all unsupported areas, using the existing features to ensure that there are no islands (unsupported areas).
- Save the file as a new Photon file.

As the user will probably have very limited options to assist in the support design and generation, it must be implemented to provide a very good support for many items. If this can be implemented, it would make adding support structure a lot easier for many users, and provide a simplified workflow.

jukefoxer, 2020-02-27:
Ok, that explains why I couldn't find the appropriate method in the PhotonsHeader class. No problem, I'm fine with just photon files.

My workflow currently is:
- I place the model in Chitubox and rotate it, so that only platform supports are needed and the big surfaces are not parallel to the build plate and export it as a photon file.
- I open the file in the PhotonFileValidator.
- I press the support button, maybe adapt some of the parameter in the dialog, and then press start.
- I save the new photon file and put it on my printer.

The default settings in the current support dialog work well for different resins that I tried with the usual exposure settings. Maybe the user has to increase the exposure time or the contact area after the first print if the supports were ripped off, or decrease the exposure time/contact area if the supports were too hard to remove. I put an explanation of this into the support dialog. 
[image]
[image]

Of course it would be nice to be able to see which support pillar locations are suggested and be able to remove some, before they are generated. I didn't have the need for this feature until now, so I didn't implement it.

How the generation of the support basically works:
1. The model is lifted by the requested distance by adding new layers on the bottom (and adjusting the exposure times accordingly).
2. The projection of the model on the XY plane is calculated.
3. The locations of the support pillars are determined. Basically it's a regularly spaced grid, but it will leave out pillars if they come too close to the model from the side (leave out pillars that would probably stick to the model from the side).
4. The optimal position of the small contact shape on the pillar is calculated (in the drawing it's always in the center, but the implementation this to move it to an ideal position on top of the pillar). The position where the contact is on the pillar is determined so that the contact area to the model is maximized.
5. The pillars are built up layer by layer.
6. The layers are recalculated and checked for islands. If there are unsupported islands the user is notified and recommended to run the fixing feature.

It's hard for me to say if that support feature is useful for other users in it's current state, because I have my specific applications and models, for which it works, and don't know how well it would work for other people. I'm pretty sure that some people won't like it because it uses more support material than the common support structures. Some may also not care too much about support marks on their models or want more than just the platform supports (i.e. inner supports).