Photonsters/PhotonFileValidator #66: .pw0 Photon Zero Support

Gainiac, 2020-04-02:
It would be great if the PhotoFileValidator could support the newly released Photon Zero.

Thanks a lot!

thekillermushroom, 2020-05-20:
please this would help so much

X3msnake, 2020-05-21:
does Chitubox outputs PW0?

If it does you can slice to photon, fix in validator, open the fixed file
in chitubox and resave as pw0

20/05/2020 à(s) 03:50:

paszkoski, 2020-11-25:
That would be great!

draon7, 2020-12-02:
> […](#)
> -- Com os melhores cumprimentos, Vinicius Silva

Chitubox unfortunately does not export to pw0 yet, but they wanted to release it with the last version so we hopefully get it soon. Even though it would be much better workflow if photon file validator would support it directly. Not speaking of much lower dpi of the screen and therefore possibility for creation of islands.