Photonsters/PhotonFileValidator #71: Problems printing with chitubox and elegoo mars

Boozebeard, 2020-05-02:
I was trying to print the model shown below

[image]

But kept getting flat prints like this

[image]

During my searches I saw mention of file validator and chitubox not working well together some times. I now realise that this was supposedly fixed in 2.1 (I am using 2.1) however before seeing this I tried printing the file without running it through PFV and it finally worked... so is there a new issue here? 

I have just recently installed the 1.6.4.3 beta. I definitely used the stable 1.6.3 for at least one of the prints but maybe that has something to do with it?

X3msnake, 2020-05-03:
did you cycle trough the file to see if the texture was there after
resaving?

Boozebeard, 2020-05-13:
Yes, I loaded the sliced file back into chitubox and it looked fine.

X3msnake, 2020-05-13:
Is the texture is not getting stuck to the vat?

à(s) 20:18:

Boozebeard, 2020-05-13:
No, I was at a loss for what the problem was at first, so I cleaned out the vat and re leveled the print bed a few times and never noticed such a thing. I don't think it was a print setting issue as it printed fine with the same settings when I didn't pass it through file validator. Unless validator changed the settings somehow. I will try printing it again soon and double check all the settings are the same when I save it out of validator.

PollockT, 2020-05-20:
Have you done actual measurements against the three you made and the one supposed to be printed to see if the slices are coming through and are possibly an exposure issue?

closed 2021-10-12 by X3msnake