Photonsters/PhotonFileValidator #72: File does not load

fernandobelem, 2020-05-04:
Hi, 

I've tried to load multiple .chitubox/.photons files but the file does not load. 

A message shows in screen. "Complete..." like the pic below:

[image]

I've executed the jar in console prompt in order to see some exception, and I got this after select a file to be loaded:

[image]

X3msnake, 2020-05-04:
.chitubox is not a sliced file
.photons is jot supported yet.

try .cbddlp or .photon

fernandobelem, 2020-05-04:
> […](#)
> -- Com os melhores cumprimentos, Vinicius Silva

Thank you for your reply. 

Sorry for trying to load a .chitubox file.

But about the .photons file, it is a sliced file from chitubox to the Anycubic Photon S printer.  

So, when you have some time, it would be cool to PhotonFileValidation to support this kind of file.

Thank you !

X3msnake, 2020-05-05:
There is a Pull Request with lots of changes that the devs are working on,
nt sure if the PhotonS is already supported there yet.

As a work-around you can slice to cbddlp, validate with the validator then
open the sliced file in chitubox and save as .photons or use a conversion
tool like the ones listed on photonsters.org community projects

4/05/2020 à(s) 21:33:

tobecwb, 2020-05-05:
same problem here...
already tried with .cbddlp and .photons files

Sliced with Chitubox v1.6.3
The #73 issue is the same thing...