Photonsters/PhotonFileValidator #73: Perpetual "Loading file..."

soccerball40k, 2020-05-05:
I have tried to load in .cbddlp / .photon files into Photon File Validator several times and I only end up with: 
[image]

I have tried leaving it for more than an hour just to see if it was a file size issue but still no luck.
File sizes:
.photon file = 83.1 MB
.cbddlp file = 83.1 MB

[image]

I am completely new; so this may just be myself not knowing that I need to download "XYZ" update or something along those lines.

jedikalimero, 2020-05-27:
Not the same issue. #72 is about unsupported formats (.chitubox and .photons)

I have the same problem as soccerball40k in Photon File Validator 2.1.

I can load the example Anycubic mesh cube that comes in the pendrive with the Anycubic Photon but can't load any of the .photon files generated with Chitubox for the Anycubic Photon despite the Printer reads and prints them without any problem.

So there seems to be some issue with the files generated by Chitubox but it seems the fault is not in Chitubox since the printer can process them correctly.

Here is the famous twisted rook in a .photon file generated by Chitubox that doesn't load in Photon File Validator 2.1 but prints correctly in the Anycubic Photon:

[rook_test.zip](https://github.com/Photonsters/PhotonFileValidator/files/4690024/rook_test.zip)

soccerball40k, 2020-05-27:
@jedikalimero 

So I've since found my solution. Re-download the drivers for your Photon and install them. Once I did this it fixed the issue.

diezmon, 2020-05-29:
I've got the same issue, if I load anything with AA > 2.   I've tried with sliced files from Chitubox and the photon slicer.

jedikalimero, 2020-05-29:
> 
> 
> @jedikalimero
> 
> So I've since found my solution. Re-download the firm-ware drivers for your Photon and install them on the printer. Once I did this it fixed the issue.

I can't see how updating the firmware of the printer will have any influence in a program running in my computer.

The problem here is either Chitubox is not generating a correctly formed file or Photon File Validator is unable to read a correctly formed file generated by Chitubox. It has nothing to do with the printer hardware or firmware. In fact, the printer is able to read without any problem the files that Photon File Validator is unable to read.

Since the Anycubic Photon printer is the only hardware in the world that uses a .photon file and it reads the files generated by Chitubox without any problem, I would say the bug is in Photon File Validator.

diezmon, 2020-05-29:
> @jedikalimero
> 
> So I've since found my solution. Re-download the firm-ware drivers for your Photon and install them on the printer. Once I did this it fixed the issue.

yeah, that makes no sense.

jedikalimero, 2020-05-29:
> 
> 
> I've got the same issue, if I load anything with AA > 2. I've tried with sliced files from Chitubox and the photon slicer.

So have you verified that a .photon file generated by Chitubox with no Antialiasing or an Antialiasing of 2 loads correctly but with Antialiasing of 4 or 8 does not?

diezmon, 2020-05-29:
> 
> 
> > I've got the same issue, if I load anything with AA > 2. I've tried with sliced files from Chitubox and the photon slicer.
> 
> So have you verified that a .photon file generated by Chitubox with no Antialiasing or an Antialiasing of 2 loads correctly but with Antialiasing of 4 or 8 does not?

here's what I've found 

both chitu and photon slicer files load fine if AA is off, or set to 2.   

AA set to 4 from chitu didn't load. I didn't actually try 4 with photon slicer 

AA of 8 fails to load from either slicer.

soccerball40k, 2020-05-29:
> > @jedikalimero
> > So I've since found my solution. Re-download the firm-ware drivers for your Photon and install them on the printer. Once I did this it fixed the issue.
> 
> yeah, that makes no sense.

Derp, apologies I've been working 60+ hours with overnights and responded to this thread when I was thinking of another.