Photonsters/PhotonFileValidator #74: Issues loading photon file validator

Wizard83, 2020-05-24:
i am on Win10 and i have issues running photon file validator,
I install both version of win installer but it cannot load. When i click on the photon file validator, it just load the big image in the middle of screen and then nothing is loaded.
java version 8 (update 251) is installed.
Anywhere for me to check the logs on what went wrong?

i am already running on administrator rights, i have also right click and run as administrator but its not working at all.

Anyone can share some light to me what went wrong ?

HallowedTruth, 2020-12-11:
i have same problem did you ever find a solution?

X3msnake, 2020-12-13:
Use UVTOOLS instead. PFV is no longer being developed