Photonsters/PhotonFileValidator #77: Not opening Chitubox Files

d1designs, 2020-06-06:
Install appeared to be done correctly but Chitubox will not open.

Nothing works.

DrStalker, 2020-07-16:
Are you saving as the newer .ctb (Elegoo MArs Pro) the older .cbddlp (Elegoo Mars) or another format?

From my testing Photon Validator cannot open .ctb files but can open .cbddlp

derpicknicker1, 2020-07-31:
I second this. `*.ctb` does not work

JYGomer, 2020-08-08:
How do you save it as a .cbddlp file type? The only option when saving is .ctb.

X3msnake, 2020-08-08:
use UVTOOLS by Master photonster Sn4k3 instead

validator development is halted and will probably be deprecated for uvtools

X3msnake, 2021-10-12:
## PHOTON FILE VIEWER IS NO LONGER MANTAINED OR DEVELOPED!

Use [UVTools](https://github.com/sn4k3/UVtools/blob/master/README.md) by Master Photonster Tiago Conceição instead