Photonsters/PhotonFileValidator #81: All files showing up as weird polygons only

FinnAgainZero, 2020-07-15:
Hi there, I wasn't able to find any answers to this online or in the previous questions, but perhaps I missed something. 

TLDR: I've updated Java, I've tried multiple different files, but each and every one has a blank black field under the "print layer " tab and a weird geometric figure in the preview tab. 
[image]

JYGomer, 2020-08-08:
Has there been any fix to this issue? I am getting the same problem.

X3msnake, 2020-08-08:
for validation and fix try UVTools by Sn4k3 instead
for 3D visualization use the 3D viewer from photonsters.org or just open
the sliced file on chitubox

X3msnake, 2021-10-12:
## PHOTON FILE VIEWER IS NO LONGER MANTAINED OR DEVELOPED!

Use [UVTools](https://github.com/sn4k3/UVtools/blob/master/README.md) by Master Photonster Tiago Conceição instead

closed 2021-10-12 by X3msnake