Photonsters/PhotonFileValidator #95: Skipping Support Layers

jdupras, 2021-03-08:
Using the Photo File Validator on OSX. I have multiple exports on multiple versions of slicers and keep seeing the same issue. The validator seems to skip supports on certain layers and resume them, which creates an island.

[image]
[image]
[image]

jdupras, 2021-03-08:
The are auto supports generated by Prusa Slicer 2.3 and sliced on both Chitubox and Lychee shows the same result.

X3msnake, 2021-03-10:
use Uvtools instead PFV is no longer actively developed