Photonsters/PhotonFileValidator #96: OSX Install Issue

dizzwold, 2021-05-04:
Hi Guys,

 I'm trying to download this wonderful software, but keep getting an error;

The following disk images couldn't be opened
osx-photonfilevalidator no mountable file system

 Any advice would be welcome.
 I've seen this software on youtube an looks great.
 I'd be grateful for some help.
Running osx 10.11.6
Regards,
 Dizzwold.

helmarw, 2021-05-04:
cannot see any issues with the dmg-files from latest release 2.1
either with or without jre are working for me. (macOS 11.2.3)
10.11.6 is El Capitan?! thats two years older than the latest release of the PhotonFileValidator
that might be an issue, but i cant test it since i dont own such an old system anymore.
Cant you update to at least (High)Sierra or Mojave ?

i know it runs on Mojave and Big Sur, i think even on High Sierra but im not exactly sure, since its to long ago

dizzwold, 2021-05-04:
Hi Helmarw,
 No, can't update as it's an old 2009 imac.
Dizzwold.

helmarw, 2021-05-04:
iMac beginning (9.1) or end (10.1) 2009 ?

you can try install/update java and use the jar-version of the program
last resort would be to run it in virtualbox on a windows virtual machine or bootcamp ?!
the processor is pretty old too, no matter if 9.1 or 10.1

otherwise i dont know, maybe someone else has an other idea ?!

X3msnake, 2021-05-05:
Use UVTools instead. PFV is no longer developed and UVTools has all PFV features and plus

closed 2021-05-05 by X3msnake

helmarw, 2021-05-06:
thnx i didnt know that ! :)

dizzwold, 2021-05-06:
Hi Guys,

 Thank you for your input.

 I'm on a 2009 9.1 High Sierra.

 The UV Tools looks promising, but I can't see how to install it?

 I have the Prusa slicer installed, downloaded the UV tools zip and opened it (UV Tools Master file), but then what?

Dizzwold.

dizzwold, 2021-05-07:
Hi Guys,

 I'd be grateful for some help with installing the UV Tools on my mac.

 The installation instruction are a bit vague for Mac.

Dizzwold

helmarw, 2021-05-07:
did you download from the correct repository ?
https://github.com/sn4k3/UVtools
which zip did you download? the repository or release ? you need the realese UVtools_osx-x64_v2.10.0.zip!
why dont you ask at UVTools?

dizzwold, 2021-05-07:
Sorry,

 I'll ask on UVTools. I forgot what page I was on.

 Thank you guys for your input and information.

 Very grateful.

 Dizzwold